makerbase-mks/MKS-TOOL #6: MKS tft35 missing pictures.

CBDesignS, 2020-08-31:
I have just used the online version of mks-tool as it seems to be newer than the windows stand alone compiled version.
It seems to be missing up to 12 different logo/pictures to make a complete set for the tft35 runing on my sgen_l v1

these seem to be missing:-
bmp_filament.bin
bmp_enable.bin
bmp_circle_sel.bin
bmp_circle.bin
bmp_calibrate.bin
bmp_babystep005.bin
bmp_babystep0_1.bin
bmp_babystep00_1.bin
bmp_babystep.bin
bmp_arrow.bin
bmp_machine_para.bin
bmp_opt_temp.bin

these are also missing but I do not think ther are used ?

bmp_pic70x28.bin
bmp_pic70x28_1.bin
bmp_pic70x28_2.bin
bmp_pic70x40.bin

Importantly if you just delete the mks_pic folder and copy the new one over you will end up with a messy screen as there is no bmp_arrow.bin and that file seems to be a system wide arrow for returning in txt only menus and the bmp_circle.bin seems to be for highlighting the selected item in the text only menu. I will upload an image soon showing the missing items and problems.

CBDesignS, 2020-08-31:
[image]
as you can see all available images are selected and when uploaded a few seem to be missing.
[image]
[image]
[image]
[image]
[image]

makerbase-mks, 2020-09-01:
Can you provide your pic folder to me and i will test it on my TFT35?

CBDesignS, 2020-09-01:
I have decided to just buy a btt tft35 and use that because this tft just has too many problems for me. . bad 1.0.7 firmware, bad pictures in menu that randomly corrupt. I will sell the mks tft35 and let somone else have the problems. 

You do not need my picture set. just use any available then save the results and file compaire the mks_pic folders to see what files are missing between the tool version and the firmware release version.

CBDesignS, 2020-09-02:
[mksbin-classic.zip](https://github.com/makerbase-mks/MKS-TOOL/files/5162825/mksbin-classic.zip)
[mksbin-simple.zip](https://github.com/makerbase-mks/MKS-TOOL/files/5162826/mksbin-simple.zip)

both versions only have 144 bmp.bin files but the original firmware has 149 so your generated files are missing the system files that should not be changed as they are for the text menus.

Pre Heat Close img is linked to Fan 0% and they should not be the same .