makerbase-mks/MKS-TOOL #8: TFT35 V2 no fonts and corrupted pictures

ibra87280, 2020-11-04:
HI,

I have received my brand new Nano board and TFT35 screen to install on my Geeetech A10.
It was working fine until I decided to try the Web MKSTOOL to make a custom firmware for the printer.
After uploading the firmware directly with the TF Card using USB to power the board, I got fonts missing so I tried again to reupload but nothing changes.
I tried also to flash the screen only, using  Web MKSTOOL and got images corrupted.
I decided to try with the MKSTOOL executable for Windows, nothing changes.
I tried with the images files provided in the GitHub, same problem.
When the card boots to flash, I have a black screen for 10-20 seconds, so I suppose it's flashing.
If I plug the TF Card on PC, I found a folder called "bak_pic" so I suppose it does a backup of old pictures when flashing.
Here's a video showing the problem:
https://youtu.be/gG3-D8ng21M

Thanks for any help.

anteyru, 2020-11-23:
Tried to compile Marlin to support uart on the mks robin nano v2 board on the site: https://www.baizhongyun.cn/home/mkstoolview
Marlin is loading, but pictures are not installed.
In this https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X/tree/master/Firmware, the firmware is installed here normally, but all motors are inverted.
[image]
[image]

Mglez, 2021-02-07:
I have the same issue, firmware works and it is installed but no pictures in the menu. I have already put the "assets" folder with the content in the SD together with the bin file. In my case all the motors are inverted too.

duder56, 2021-02-17:
I had the same issue, you have to split the font and icons into 2 seperate folders named mks_pic and mks_font out of the assets folder, or at least this was the only way I could get it to work. that being said, MSKTOOL seems to be broken in that any changes I make don't register, such as inverting the steppers or the endstops. But maybe you'll have better luck.

Mikelodoncz, 2023-04-05:
Hello guys, I am trying to configure Marlin  for mks robin tft35 and uart, but when I define tft_lvg_ui compile it and put it on sd card with along side with its "assets" it shows this screen [image]

Am I configuring Marlin wrong? Or what, because I don't know what else to change.