makerbase-mks/MKS-TOOL #9: Marlin 2.0.7.2

anteyru, 2020-11-19:
The current version of Marlene is already out of date. Planning to update Marlin to version 2.0.7.2?
Also, there is no UART support for Robin Nano 2.

mks-viva, 2020-11-23:
Hi! If you download offical marlin 2.0.7.2 and MKS Robin Nano V2 uart mode iwork is ok?

anteyru, 2020-11-23:
Yes, everything works fine. Therefore, I am surprised that you have a very old version.

mks-viva, 2020-11-23:
MKS TOOL may not be updated yet, it is recommended that you directly use the official firmware.

anteyru, 2020-11-23:
Marlin 2.0.7.2
[image]

mks-viva, 2020-11-23:
Wa, look it is very ok!

anteyru, 2020-11-23:
Firmware from MKSTOOL https://www.baizhongyun.cn/home/mkstoolview
[image]
[image]

mks-viva, 2020-11-24:
This looks a bit bad.

anteyru, 2020-11-24:
There is no uart in the official firmware, the motors turn in the other direction.

mks-viva, 2020-11-24:
Uart settings is easy.
1. In Configuration.h file set driver stype as: TMC2209
(If use TMC2208 or TMC2225, you can set it directly)

anteyru, 2020-11-24:
I'm talking about the standard firmware.
makerbase-mks/MKS-Robin-Nano-V2.X/Firmware/Robin-nano-V2.0-Marlin-bugfix2.0.x-MKS-2.1.1.rar

mks-viva, 2020-11-25:
Can you provide a vodies for me when you use makerbase-mks/MKS-Robin-Nano-V2.X/Firmware/Robin-nano-V2.0-Marlin-bugfix2.0.x-MKS-2.1.1.rar firmware？

anteyru, 2020-11-26:
> Can you provide a vodies for me when you use makerbase-mks/MKS-Robin-Nano-V2.X/Firmware/Robin-nano-V2.0-Marlin-bugfix2.0.x-MKS-2.1.1.rar firmware？
Home and zero to the left near corner.
Clicking on "+" moves to 0.
https://www.youtube.com/watch?v=TbBSxuNS5eA

anteyru, 2020-11-26:
[Robin Nano v2 motor direct](https://www.youtube.com/watch?v=TbBSxuNS5eA)

mks-viva, 2020-11-26:
> > Can you provide a vodies for me when you use makerbase-mks/MKS-Robin-Nano-V2.X/Firmware/Robin-nano-V2.0-Marlin-bugfix2.0.x-MKS-2.1.1.rar firmware？
> > Home and zero to the left near corner.
> > Clicking on "+" moves to 0.
> > There is no way to change direction from the menu.
> > If you collect through the site, the firmware is configured. But the pictures are not loaded.
> > [MKS-Robin-Nano-V2 motor direct](https://www.youtube.com/watch?v=TbBSxuNS5eA)

You need to modify the direction parameter in Configuration.h file and recompile and reupdate it

anteyru, 2020-11-26:
> You need to modify the direction parameter in Configuration.h file and recompile and reupdate it

Ok, but then how do I add the GUI from ronin nano v2 to my marlin?

mks-viva, 2020-11-26:
Offical Marlin has add MKS UI for MKS Robin Nano V2.

anteyru, 2020-11-26:
> Offical Marlin has add MKS UI for MKS Robin Nano V2.

> // TFT LVGL UI
//
// Using default MKS icons and fonts from: https://git.io/JJvzK
// Just copy the 'assets' folder from the build directory to the
// root of your SD card, together with the compiled firmware.
//
//#define TFT_LVGL_UI_FSMC  // Robin nano v1.2 uses FSMC
#define TFT_LVGL_UI_SPI   // Robin nano v2.0 uses SPI

anteyru, 2020-11-26:
Is it about this?

mks-viva, 2020-11-26:
The latest version Marlin-bugfix-2.0.x . If you want use MKS UI, you need set it:
Platformio.ini file: 
```
    Set default_envs = mks_robin_nano35
```
Configuration.h file: 
```
    #define SERIAL_PORT 3
    #define MOTHERBOARD BOARD_MKS_ROBIN_NANO_V2
    #define MKS_TS35_V2_0
    #define TFT_LVGL_UI
    #define TOUCH_SCREEN
    #define SDSUPPORT
```

tizi24, 2022-10-11:
is it compatible with MKS TS24 V2.0?