makerbase-mks/MKS-TOOL #14: TFT35 Problem to personalize it with MKSTools

Magiquall, 2021-03-08:
Hello, I have ordered the TFT35 screen from you (https://fr.aliexpress.com/item/32891320510.html?spm=a2g0s.9042311.0.0.27426c371gsTVa). With card MSK SGEN_L V2.0
And I have a problem to personalize it with MKSTools and Makerbase Image Editor.

Already, MakerbaseImageEditor, is unable to read correctly and therefore extract images for the 3.5 Inches, do you plan to update this application. Because for the moment the images are duplicated, as if they were loaded in a bad resolution.
This is what happens in MKSTools, but in this last one can choose the resolution of the screen.

Regarding MKSTools, I was able to use it, but I have to choose what as a card. Because some menu is not arranged like what is on my screen and when I try to update with the files created with MKSTools, during the update my screen reboots while writing the image files ". bin ". It is therefore impossible to use it to personalize my screen.

These programs really do not seem suitable for your TFT35 screens.

I am a programmer, you have the possibility to provide me with the source codes of these 2 programs, I do not know if I would be able to improve them, but in any case for the moment they do not work as they are supposed to do.

But sincerely considering the praise that some Youtubeur carried you a few years ago, or few  months. I have found a lot of people who no longer want to use makerbase because of this kind of little problem, it's a shame.

Another little suggestion.
Your displays do not have the 12864 display emulation, but it would be nice to have at least a view of the print state, when the board is driven by an external unit, such as a Raspberry PI and Octoprint.
A view with possibly the temperature (bed and nozzle), as well as the layer number. I don't think it's complicated to have a view for just see these few parameters.

leofaure, 2021-03-13:
I totally agree

ChristopheSV, 2021-03-22:
I tried the MKS Tools site, and it's the same, it doesn't work, the images are not all replaced, and the position is not exactly the same between the software and the TFT, hence probably errors . As with PC software, at least some consistent, both do not work.
In short, it's useless, and no response from Makerbase, no follow-up, in short it's decided I'm going to BTT hoping that it is more attentive to customers, but it's not complicated, Bye.

Also now i have the screen to show the temperature values its good but when i sent the print the bed is heating up but but dont refresh the value temperature, and dont print. I tried to reload the firmware which was working, but it still does not work. It is really undesirable.

leofaure, 2021-03-22:
hola ami ya me está funcionando el mks robín nano v 2.0 con controlador tmc 2209 y wifi. el mks tool está roto totalmente dejado, y el mks Cloud para iOS no funciona, cuando se coonfigura la impresora y se conecta, la aplicación se queda con un error.

es una pena que los desarrolladores no hagan bien su trabajo con el personal necesario para dar soporte a todos los compradores de sus productos.