makerbase-mks/MKS-TOOL #21: fatal error: TMCStepper.h: No such file or directory

msoushi, 2021-07-15:
为MKS Gen_L 2.1编译开启了TMC2208的固件时 出现了缺少TMCStepper库的错误
```
Processing mega2560 (platform: atmelavr; board: megaatmega2560; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR (3.3.0) > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
DEBUG: Current (avr-stub) On-board (avr-stub, simavr)
PACKAGES:
- framework-arduino-avr 5.1.0
- toolchain-atmelavr 1.70300.191015 (7.3.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 7 compatible libraries
Scanning dependencies...
Dependency Graph
|-- 1.5.0
|-- 0.4.3
| |-- 1.0
| |-- 1.0
|-- 1.0
|-- 1.0
|-- 2.0
|-- 1.0
Building in release mode
Compiling .pio\build\mega2560\src\src\HAL\AVR\HAL.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\HAL_SPI.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\MarlinSerial.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\Servo.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\eeprom.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\fast_pwm.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\fastio.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\u8g_com_HAL_AVR_sw_spi.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\watchdog.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\HAL_spi_L6470.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\eeprom_api.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\eeprom_if_i2c.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\eeprom_if_spi.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\esp_wifi.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\servo.cpp.o
Compiling .pio\build\mega2560\src\src\MarlinCore.cpp.o
Compiling .pio\build\mega2560\src\src\core\serial.cpp.o
In file included from Marlin\src\module/stepper/indirection.h:44:0,
from Marlin\src\module/stepper.h:47,
from Marlin\src\MarlinCore.cpp:53:
Marlin\src\module/stepper/trinamic.h:29:10: fatal error: TMCStepper.h: No such file or directory

********************************************************************
* Looking for TMCStepper.h dependency? Check our library registry!
*
* CLI > platformio lib search "header:TMCStepper.h"
* Web > https://platformio.org/lib/search?query=header:TMCStepper.h
*
********************************************************************

#include
^~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\mega2560\src\src\MarlinCore.cpp.o] Error 1
========================== [FAILED] Took 8.20 seconds ==========================

Environment Status Duration
------------- -------- ------------
mega2560 FAILED 00:00:08.203
==================== 1 failed, 0 succeeded in 00:00:08.203 ====================
```

mks-viva, 2021-07-16:
把你的整个固件给我，我测试下

msoushi, 2021-07-16:
不知道为什么没法直接传到github上，传个蓝奏云试下吧
[固件](https://msoushi.lanzoui.com/ibnbErh3sch)

mks-viva, 2021-07-16:
哦豁，我电脑可以编译：
[firmware hex.zip](https://github.com/makerbase-mks/MKS-TOOL/files/6827190/firmware.hex.zip)

msoushi, 2021-07-16:
那……在线编译器的问题不考虑修一下吗？
虽然确实挺方便，但这个问题还是有点难受……

makerbase-mks, 2021-07-31:
@msoushi 已经修复了

closed 2021-07-31 by makerbase-mks