makerbase-mks/MKS-TOOL #28: Can't upload Configuration.h file (Configuration.h与所选机型不一致，请重新选择或上传)

maticlms, 2022-09-19:
When configuring TFT for MKS Robin Nano V1.2 it asks you to define the cachine (I selected Two Trees Sapphire Plus) and you have to upload Configuration.h and Configuration_adv.h files. I can upload Configuration_adv.h fine, but when trying to upload Configuration.h it says Configuration.h与所选机型不一致，请重新选择或上传.

axium77, 2022-10-03:
Same problem!!! Mine is ANET E16 ! I tray different models - no one works!

Hueristic, 2022-10-14:
Same although I tried to upload 2.11 and 2.1x configs and they failed with chinese lang error but site accepted 1.19 config files

Hueristic, 2022-10-14:
与所选机型不一致，请重新选择或上传 == "If it doesn't match the selected model, please select or upload it again" according to google.